ExpressLRS/Docs #13: Describe USE_DYNAMIC_POWER user define

jurgelenas, 2021-08-01:
From [#724](https://github.com/ExpressLRS/ExpressLRS/pull/724)

jurgelenas, 2021-08-01:
I wrote this for the configurator app:

```
Dynamic power feature uses LQ-based power boost up to detect
quick drops in link quality and RSSI-based power adjustments
which are working slowly, suitable for a general long-range
flights.

This feature requires ENABLE_TELEMETRY.
```

CapnBry, 2021-08-01:
The feature does not require ENABLE_TELEMETRY :-P I've got a doc started in the wiki for this and will get it in the website once I stop making updates.
https://github.com/ExpressLRS/ExpressLRS/wiki/Dynamic-Transmit-Power

maybenikhil, 2021-08-02:
will wait to work on this until you finish the wiki page

jurgelenas, 2021-08-02:
> The feature does not require ENABLE_TELEMETRY :-P I've got a doc started in the wiki for this and will get it in the website once I stop making updates.
> https://github.com/ExpressLRS/ExpressLRS/wiki/Dynamic-Transmit-Power

thanks, will update my description!

jurgelenas, 2021-08-18:
Follow up?

closed 2021-08-29 by maybenikhil