ExpressLRS/Docs #62: [2.0] Update Model Configuration and Matching Behavior Documentation

maybenikhil, 2021-09-09:
This requires updating software/model-config-matching.md and quick-start/lua-howto.md to be in line with the new behavior of model Matching

CapnBry, 2021-09-11:
@pkendall64 have you started this task? If not I can make a whole page about it for the docs. There's not a ton to say about it, but I think too much to cram into the modelmatch section of the Lua page
https://www.expresslrs.org/2.0/quick-start/lua-howto/#model-match

pkendall64, 2021-09-11:
No, I haven't, sorry. I did the initial cut which @maybenikhil made a lot nicer, but I think it just needs the Set RX Model removed and then a once-over the whole thing.

CapnBry, 2021-09-11:
> No, I haven't, sorry. I did the initial cut which @maybenikhil made a lot nicer, but I think it just needs the Set RX Model removed and then a once-over the whole thing.

Roger that. I just wanted to ask to make sure that we weren't going to be duplicating work. I'll make a draft PR tomorrow.

PR #66 (2021-09-25): Update for Model Match half-connected PR
Fixes #62
closed 2021-10-04 by maybenikhil