ExpressLRS/Docs #92: Add description for USE_TX_BACKPACK user define

jurgelenas, 2021-10-31:
See: https://github.com/ExpressLRS/ExpressLRS/blob/master/src/user_defines.txt#L67

CapnBry, 2021-11-03:
The `USE_ESP8266_BACKPACK` should also be removed from user-defines and added to obsolete-defines. Note that the link to obsolete-defines is broken and should link to `obsolete-defines` but links to `Obsolete-user_defines` instead so that should be fixed as well.

maybenikhil, 2021-11-04:
is this broken on both master & dev or just one?

maybenikhil, 2021-11-04:
just dev oops

CapnBry, 2021-11-04:
That wasn't for you, I was just making a note for myself when I addressed this issue. Go back to class!

maybenikhil, 2021-11-04:
i was doing this real quick haha, i'll stop now

CapnBry, 2021-11-04:
Haha you can do it if you want, but it's my define so it's on my to do list to document it. We do have a bunch of doc PRs piled up though kissy face emoji

CapnBry, 2021-11-19:
Only took me three weeks but I DID IT!

PR #113 (2021-11-22): Add USE_TX_BACKPACK
Fixes #92

Sorry sorry sorry sorry sorry sorry sorry. I completely forgot about this being on my task list. Also fixes the link to obsolete-defines.
closed 2021-11-22 by maybenikhil