ExpressLRS/Docs #102: openocd for Ghost Atto

macsleep, 2021-11-07:
I used the following openocd command on Linux to wipe an ImmersionRC Ghost Atto for ExpressLRS installation:

`openocd -f interface/stlink-v2.cfg -f target/stm32f3x.cfg -c 'init; reset halt; stm32f3x unlock 0; flash protect 0 0 last off; stm32f3x mass_erase 0; reset halt; exit'`

Maybe this is something worth adding to the [open-ocd.md](https://www.expresslrs.org/1.0/software/open-ocd/)?

maybenikhil, 2021-11-09:
I don't see why not, would you like to add it in or would you like us to add it?

macsleep, 2021-11-10:
Feel free to add it. I'm not sure how best to proceed (note, list of openocd commands per receiver etc.) and wanted to leave that up the author.

closed 2021-11-19 by maybenikhil

macsleep, 2021-11-19:
Looking good :-)