ExpressLRS/Docs #161: UID Byte Generator is down.

LoakAudio, 2022-01-19:
## Summary

Hello all.
In ExpressLRS Docs ([here](https://www.expresslrs.org/2.0/hardware/spi-receivers/#uid-byte-generator)), the UID Byte Generator seems to be down.
Sorry to probably post in the wrong place as a question but i didn't know where to post about docs.

Edit : I've just seen the "Docs Github", what a jerk, sorry , maybe an admin can move this post ...

jurgelenas, 2022-01-19:
> ## Summary
>
> Hello all. In ExpressLRS Docs ([here](https://www.expresslrs.org/2.0/hardware/spi-receivers/#uid-byte-generator)), the UID Byte Generator seems to be down. Sorry to probably post in the [...]

No worries, thanks for the bug report! I have transferred the issue to docs repository :)

jurgelenas, 2022-01-19:
@busheezy can you take a look at this, please?

busheezy, 2022-01-19:
It still seems to function for me. I would make sure you don't have JS disabled.  
  
If you still can't get it to work, can you please tell me if the following link works? 
https://busheezy.github.io/elrs-binding-phrase-to-bytes/

Thanks

LoakAudio, 2022-01-19:
Yes, it's strange, during several tests on a couple of computers, sometimes it works sometimes it doesn't. Maybe a connection problem, the web seems overloaded for a few days. Last week end Github was unreachable.
So far everything is working (the web page, and your link too)

busheezy, 2022-01-19:
The next time that it doesn't load, would you mind looking at your browser's console and see if you notice any errors?

LoakAudio, 2022-01-19:
Of Course, I've found the console in firefox, I will have a look.

closed 2022-06-01 by maybenikhil