ExpressLRS/Docs #195: GSoD'22 initial contact

zededasatyam, 2022-04-17:
Hello community,

I am Satyam Kumar, an open source enthusiast and 3rd year Computer Science and Engineering undergraduate at National Institute of Technology Hamirpur.
I have successfully completed m project for MBDyn as a developer in GSoC 2021. I wish to contribute to the documentation of the organisation this year. I have good experience of writing technical documents as I have been doing this in my previous two internships for past one year.

Please let me know how to proceed. I really hope to work with the organisation and have a great summer ahead.

maybenikhil, 2022-04-27:
Hello Satyam, if you have not already - we recommend sending a developer (@Jye, me, or deadbyte specifically) your resume and your timeline for review in the coming weeks.

zededasatyam, 2022-04-28:
Hello @maybenikhil 
I am attaching my resume here 
[resume.pdf](https://github.com/ExpressLRS/Docs/files/8581816/resume.pdf)

I have good experience of writing technical docs as in my past internships, I have written TCDs, but due to the company policy, I cannot share them at a public place.

As for my timeline, I am open to work from 7th of May 2022 onwards, as my exams will be over by then. I will be able t dedicate about 25 hours per week. I will surely put in some extra efforts whenever required.

PS: while creating the issue, I didn't notice I was logged in with my organisation's id (the one I am interning till mid May 2022). My personal GitHub id is: https://github.com/skdbsp123 and I shall continue to chat with this id from now onwards.

zededasatyam, 2022-05-06:
Hi @maybenikhil ,
Any updates?

maybenikhil, 2022-05-06:
Hello, we are currently reviewing and selecting applicants at the moment. We will send discord/email follow ups to all applicants once we make a final pick

closed 2022-05-24 by maybenikhil