ExpressLRS/Docs #196: Update/Fix naming conventions

commanderguy3001, 2022-04-17:
Currently naming conventions don't exist/are unintuitive/aren't followed/don't make sense.

One example to name directly:
https://expresslrs.org/2.0/quick-start/transmitters/tx-diy
vs
https://expresslrs.org/2.0/quick-start/receivers/rx-diyreceivers/
should be:
https://expresslrs.org/2.0/quick-start/transmitters/diy
and
https://expresslrs.org/2.0/quick-start/receivers/diy

Notice the differences:
- the names for receivers and transmitters are the same, as it makes sense (`diy` for the DIY receivers, as well as the DIY transmitters)
- the `tx-` and `rx-` part got removed, as it doesn't make sense to have it, as it is redundant information (we already have the `transmitters/` and `receivers/` parts)

A couple other things that bother me:
- the `Flashing Receivers` and `Flashing TX Modules` aren't coherent in their naming, should either be `Flashing Receivers` and `Flashing Transmitters` OR `Flashing RX Modules` and `Flashing TX modules` (I reckon the former one is easier to understand)
- All the documented Modules have a `Flashing <module>` before their name which makes it quite a bit harder to read, remove that as we know that we are flashing stuff and it's redundant information
- The documented Modules in the List on the website aren't sorted, which makes it really hard to find the exact one you wanna flash right now, as you gotta look at every single line, when it would be sorted from A-Z this would be easier
- SPI based receivers aren't mentioned as a receiver model, might be nice to have a `SPI based AIO flight computers` section there too, and reference the /hardware/SPI-receivers section.

deadbytefpv, 2022-04-28:
#197 has been merged.
Could you reevaluate the site and post anything that bothers you still?

commanderguy3001, 2022-04-28:
yep, looks a lot better now, only things that still bother me:
- the lists aren't sorted A-Z (well, they are sorted until Vantac, but after that there's chaos)
- instead of having `Siyi TXes` I think `Siyi FM30` might be a better choice
- for happymodel EP and frsky R9 there's a small `s` after the item in the list, but not for `Matek R24`, I think putting a `*` behind all of these might work the best, but that's something I don't know for sure
- - also for namimno flash RXs there are multiple versions but no `s` behind them
- shouldn't it be `Frsky Vantac` (I thought vantac is by frsky?)
- similar for ghost, I think adding `ImmersionRC` would be better suited
- some devices (ghost is one notable) don't really have a proper documentation, but that's probably a good idea to put into a new issue

maybenikhil, 2022-06-06:
Looks better now from #213 . I think we said Vantac b/c frsky doesn't have their name on it. Some still may be changed but generally formatting is better

closed 2022-06-06 by maybenikhil