ExpressLRS/Docs #258: New Receivers for selection page.

FrenzyIncarnate, 2022-10-16:
Hello. 
There are some new receivers that I've seen on the market that are not in [the receiver selection page ](https://www.expresslrs.org/3.0/hardware/receiver-selection/) 
I have gathered information for 2 of them and you can find it below. 
[elrs new receivers.xlsx](https://github.com/ExpressLRS/Docs/files/9794403/elrs.new.receivers.xlsx)
[image]

Thank you for your good work.

maybenikhil, 2022-11-14:
Sorry for the slow reply - we need to do a more broad overhaul of that page in general which is on my list of things to do. Thanks for bringing it up, I appreciate the contribution

BhaswatiRoy, 2023-01-16:
Hello, we can update the new receiver details in the receiver-selection.md file
I can work on this!

maybenikhil, 2023-01-18:
In theory i'd like to change it to more of a feature guide than a list of every receiver because there are so many now, but I am not sure what my timeline for that is so if you'd like to add these in for the time being that would be alright as well

BhaswatiRoy, 2023-01-19:
Hello @maybenikhil 
I have added the 2 new receivers in the receiver-selection.md file.

closed 2023-01-20 by maybenikhil